cicciocb/Annex32_examples #21: Implement the pin drive capability

cicciocb, 2024-05-16:
This is a feature available for the ESP

typedef enum {
    GPIO_DRIVE_CAP_0       = 0,    /*!< Pad drive capability: weak          */
    GPIO_DRIVE_CAP_1       = 1,    /*!< Pad drive capability: stronger      */
    GPIO_DRIVE_CAP_2       = 2,    /*!< Pad drive capability: medium */
    GPIO_DRIVE_CAP_DEFAULT = 2,    /*!< Pad drive capability: medium */
    GPIO_DRIVE_CAP_3       = 3,    /*!< Pad drive capability: strongest     */
    GPIO_DRIVE_CAP_MAX,
} gpio_drive_cap_t;

gpio_set_drive_capability((gpio_num_t)pin, (gpio_drive_cap_t)value );

cicciocb, 2024-05-16:
Implemented the function 

PIN.STRENGTH pin_nb, drive_cap

DB9JG, 2024-05-17:
This feature could be the very last resort for a hardware issue with an ESP32 driving a peripheral component that needs unexpected amount of current

cicciocb, 2024-05-17:
This has been implemented in the version 1.53.3.
In fact I found the interest into implementing this feature for exactly the opposite reason.
The reason ? The noise, a pin with too much strength that carries an high frequency signal generates a lot of EMI, in particular on the board itself. 
I found this issue when I designed a PCB for the VGA, solved reducing the strength from 2 (the default) to 0

DB9JG, 2024-05-17:
Well, that is interesting indeed! 
I'll have to keep that in mind for future layouts and/or possible talkover between parallel wires. 
Thanks!

DB9JG, 2024-05-17:
Forgot to introduce myself: DB9JG==PeterN
But I guess you already knew :-))

cicciocb, 2024-05-20:
Implemented the function in the version 1.53.3

PIN.STRENGTH pin_nb, drive_cap

closed 2024-05-20 by cicciocb

cicciocb, 2024-05-23:
The help must be updated