cicciocb/Annex32_examples #30: DSP / FFT

cicciocb, 2024-06-01:
Implement functionalities DSP / FFT

Zusi256, 2024-08-27:
Hi cicciocb,  i am really in need of the whole DSP/FFT functions as im now using the RP2040 as a math coprocessor :-)
but its obvious, nearly nobody uses them, when they are not implemented. ive no idea about the additional programming space which are used by the whole bunch of  DSP, maybe it could be done as a different version like the BLE.
thanks a lot for your excellent work!

By the way, an idea ist to split up the tft objects  in different pages, and to have also the ability to use the mouse as an event-catcher for the TFT display (would be nice for tft without touch).

cicciocb, 2024-08-27:
HI, the best is to write on the forum and explain what is your need and what you would to do with the FFS/DSP. Maybe you'll find someone else interested ....